bancika/diy-layout-creator #926: Single coil pickup limits?

FlounderJr, 2025-05-14:
Working on replicating a guitar layout based on a vintage Japanese model. The layout has multiple switches and 4 single coil pickups. When I run Analyze Guitar Diagram, pickup #1 is never included. Everything looks good as far as 2, 3 and 4 in the analysis. I have modified the wiring several times without success. Is there a restriction on the number of pickups?

bancika, 2025-05-15:
there's no limit. I suggest fiddling the leads of the first pickup on both
sides the make sure they join exactly where they should in order  to make
contact. If it still doesn't work you can upload the diy file here (rename
to .txt first) for me to check.

FlounderJr, 2025-05-15:
I appreciate your help. 

[Enclosed Rotary Test Teisci K4L.txt](https://github.com/user-attachments/files/20232160/Enclosed.Rotary.Test.Teisci.K4L.txt)

bancika, 2025-05-16:
In 5,5.0 I'll make the whole copper pad conductive and much more forgiving...for now you'll just need to move it for a few pixels

closed 2025-05-16 by bancika