bancika/diy-layout-creator #930: smaller or resizable pot symbol variant for schematics

topstonemusic, 2025-05-18:
Is it possible to have a smaller potentiometer schematic symbol variant (or else to make the current symbol resizable)? 
Sorry if I'm posting in the wrong place, if the request has already been made, &/or if I'm missing something... it's my first post!
   ~Christopher

topstonemusic, 2025-05-18:
I'm getting used to the grid & it's settings, but still a newbie even ~25+ hours using the software, so trying to make picture-perfect schematics for guitar effects requires lots of toggling between snap-to-grid & not. I'm on Mac OS 10.15 Catalina, fwiw.

bancika, 2025-05-18:
Not sure I understand how the two posts are related. Can you please
elaborate the need for a smaller resistor. Also, what's the issue with
snapping?

topstonemusic, 2025-05-18:
Because there's only 1 size potentiometer for drawing schematics, it sort of dictates the scale of the rest of the schematic.
Instead, in my stubbornness, I've scaled down all other components I use. 

[image]

It's a minor aesthetic issue that I'm requesting a smaller potentiometer symbol. Ideally I'd like to make every schematic symbol whatever size I wish(!). In order to unify all the component sizes a bit more nicely than in my sample, I will work under the assumption that I can only start to make my resistors & cap symbols larger, because some symbols like a BJT transistor & the potentiometer have set sizes.

Also, now I see the *snap to grid icon* seemingly for the first time. Sorry, I'd been using the menu item only. Would it be possible to have a keyboard shortcut to toggle its state?

Thank you!
~Christopher

bancika, 2025-05-18:
Already have the keyboard shortcut...please refer to the manual or the ai
asistent