bancika/diy-layout-creator #934: Pot wiring discrepancy

FlounderJr, 2025-05-19:
Analyze Guitar shows "volume.1-2" and "volume.2-3" in the circuit. It also states that the volume pot is wired in reverse.

There is only one volume pot so I don't know what the two volumes are in the analysis, and I'm pretty sure the one volume pot is wired correctly. 

[Fixed Teisco K4L wiring.txt](https://github.com/user-attachments/files/20302213/Fixed.Teisco.K4L.wiring.txt)

bancika, 2025-05-19:
each pot consists of three terminals (1-2-3) and two variable resistors
(1-2 and 2-3)

as for the reversing...it seems to be a bug, something about the tone
control confuses it. When I remove it, it detects fine. Will need further
investigation.

bancika, 2025-05-19:
found the issue - your volume pot is not grounded...once I added a lead to
the tone pot back, it started working fine

closed 2025-05-19 by bancika

FlounderJr, 2025-05-19:
The pot is actually grounded by virtue of being attached to an aluminum scratch plate that is grounded. In diagramming the original wiring of the guitar, I hadn't considered that.