bancika/diy-layout-creator #938: 'X' character not displayed in email input & no registration confirmation email

Romano-the-one, 2025-05-21:
When trying to enter an email address containing the letter 'X'  into an input field (e.g., during cloud registration), the 'X' character does not appear. It seems to be silently ignored or filtered out.

Furthermore, after attempting to register (likely with an email address that was intended to contain an 'X' or even trying with a different one if the 'X' issue was noticed), no confirmation email is received to complete the cloud account signup.

bancika, 2025-05-21:
It will be fixed soon. In the meantime, please use copy-paste into those
fields

StateOfEpicicity, 2025-05-23:
This is also happening in the name field for other objects like pots and wrap-around text labels.

bancika, 2025-05-23:
yes yes, all is fixed

closed 2025-05-28 by bancika