bancika/diy-layout-creator #946: Extra load not recognized by Guitar Analyzer

StateOfEpicicity, 2025-06-03:
There used to be a guide for how to report bugs, including where to find the log files, but I don't see that here, so maybe they're not necessary anymore with recent updates.  I did try to search for them in the Program Files folder, but couldn't find them.

Anyway, before the Guitar Analyzer was revamped, it would list when extra caps or resistors were added to a part of the diagram, but it seems not to be doing that anymore.  I'm uploading my latest diagram, which adds an extra resistor in position 2 to darken the load and simulate something closer to modern wiring, wheras I'm using straight 50s wiring in the rest.  Anyway, that extra resistor is not recognized by the Guitar Analyzer.  Paradoxically, the AI Circuit Analyzer did recognize it!

Thanks as always!

[HH 5-Way VV Presets, 50s & Modern.zip](https://github.com/user-attachments/files/20580476/HH.5-Way.VV.Presets.50s.Modern.zip)

StateOfEpicicity, 2025-06-10:
Thanks for taking a look!  Sorry, the dual gang pots are not important for what I was trying to point out.  I should've left them off.  The only thing I wanted to point out was that, in position 2, that 470K resistor is not being recognized by the guitar analyzer.  The idea there was to provide extra load on the pickup for that position only.  The idea in position is that you keep a higher load on the pickup throughout, emulating something like what Modern Les Paul wiring would give you.  I tried it in practice, and I didn't like the result, so I ditched this idea, but still, the old versions of DIYLC would've understood the resistor to be in the path in position 2; that's really all I was trying to say.  😊

If you're curious, the idea behind the dual gangs is to create 50s Les Paul wiring for each knob position, with low value caps, and for the tone pot, which is pictured in the diagram as the bottom pot, to follow the volume pot completely.  I used to use dual concentric instead of dual gang for this purpose.  With dual concentric pots, I found that I would always end up finding that the best, most consistent tone ended up being the result of moving the tone knob at the same rate as the volume knob.  So I thought, why even have a different tone control, when it would be much more convenient to have one pot shaft control two different pots, which is what a dual gang gives you!  This works functionally for me as if it were the ideal treble bleed.

StateOfEpicicity, 2025-06-10:
[HH 3-Way Resistor Example.zip](https://github.com/user-attachments/files/20680516/HH.3-Way.Resistor.Example.zip)

Sorry, I completely forgot to include a new file!  Here's a stripped down diagram, just one pickup engaged.  In position 2, it should recognize that there's an extra resistor loading the pickup.

bancika, 2025-06-10:
it works for me when I fixed the wiring on the switch

[image]

StateOfEpicicity, 2025-06-10:
You're totally right.  I mixed up where the common was on the 3-way, and that prompted me to have another look at my original diagram, and it's my mistake entirely.  🤦‍♂️  You're analyzer was trying to tell me I hadn't drawn my diagram correctly, but I was insisting I was right instead!  Shit.  Sorry Man, and thanks for taking a look!

closed 2025-06-11 by bancika