bancika/diy-layout-creator #963: [Feature Request] Wiring Layout

XTRXTRPI, 2025-09-25:
I like the current wiring options 2,3,4,5...etc curve points this is great for many types of wiring like point to point turret or component inside of a handwired amplifier.

I request that in addition there be a feature to use a mouse to drag Y to point, click drag x direction to point, click drag Y again etc. with a radius corner, such that each click represents a 90° or 45° corner and wiring can feel more like a pencil draw around components especially when multiple wires are going in the same general end point.

This would clean up some crossing of wires if wanted, in some cases wires need to be crossed at 90° angle to reduce to minimum capacitive coupling. In general it would cleanup the drawing of a layout making it easier to read and follow.

Thank you for your great app.

bancika, 2025-12-26:
Hi @XTRXTRPI , I'm not sure I understand the request. Do you have any examples?

bancika, 2026-09-13:
It took me a year but I think I understand it. The next version will include something along those lines, right angle wires that go only horizontally or vertically and can bend two or more times at right angles, with rounded corners.

bancika, 2026-10-15:
@XTRXTRPI ortogonal wires have been deployed. Check them out

closed 2026-10-15 by bancika